start-jsk/hrpsys #139: /opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/longfloor.wrl cannot be found.

130s, 2015-03-12:
I confirmed this happens on 2 different Hydro machines, only on simulators.

```
$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch 
:
respawning...
[hrpsys-3] restarting process
process[hrpsys-3]: started with pid [5346]
loading file:///opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/longfloor.wrl
/opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/longfloor.wrl cannot be found.
Retrying to load the file as a standard VRML file
/opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/longfloor.wrl cannot be found.
loading failed.
/opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/longfloor.wrl cannot be found.
/opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/longfloor.wrl cannot be found.
omniORB: ERROR -- A servant has been deleted that is still activated.
      id: root<7> (active)
respawning...
[hrpsys-3] restarting process
:
```
Versions.
```
$ dpkg -p ros-hydro-openhrp3 ros-hydro-hrpsys ros-hydro-rtmros-common ros-hydro-rtmros-hironx ros-hydro-rtmros-nextage|grep Vers
Version: 3.1.7-3precise-20150213-1403-+0000
Version: 315.3.0-1precise-20150311-1614-+0000
Version: 1.2.4-1precise-20140921-1446-+0000
Version: 1.0.24-0precise-20140921-1938-+0000
Version: 0.6.1-0precise-20150311-1850-+0000

$ find /opt/ros -iname longfloor.wrl
/opt/ros/hydro/share/OpenHRP-3.1/sample/model/longfloor.wrl
```

Doesn't happen with Trusty-Indigo.
```
$ dpkg -p ros-indigo-openhrp3 ros-indigo-hrpsys ros-indigo-rtmros-common ros-indigo-rtmros-hironx ros-indigo-rtmros-nextage|grep Vers
Version: 3.1.7-15trusty-20150213-1223-+0000
Version: 315.3.0-1trusty-20150311-1141-+0000
Version: 1.2.8-0trusty-20150311-1229-+0000
Version: 1.0.28-0trusty-20150311-1244-+0000
Version: 0.6.0-0trusty-20150311-1247-+0000
$ find /opt/ros/ -iname longfloor.wrl
/opt/ros/indigo/share/OpenHRP-3.1/sample/model/longfloor.wrl
```

130s, 2015-03-12:
Labelled as critical since this might prohibit any Hydro machine to run hrpsys simulation.

k-okada, 2015-03-14:
```
k-okada@ubuntu:~$ ls -al /opt/ros/hydro/share/openhrp3
ls: cannot access /opt/ros/hydro/share/openhrp3: No such file or directory
k-okada@ubuntu:~$ sudo apt-get install ros-hydro-openhrp3
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer
required:
  ros-hydro-rtshell ros-hydro-rtmbuild ros-hydro-rtctree
ros-hydro-cmake-modules ros-hydro-rtsprofile ros-hydro-openrtm-tools
Use 'apt-get autoremove' to remove them.
The following NEW packages will be installed:
  ros-hydro-openhrp3
0 upgraded, 1 newly installed, 0 to remove and 141 not upgraded.
Need to get 0 B/7,843 kB of archives.
After this operation, 29.1 MB of additional disk space will be used.
Selecting previously unselected package ros-hydro-openhrp3.
(Reading database ... 314574 files and directories currently installed.)
Unpacking ros-hydro-openhrp3 (from
.../ros-hydro-openhrp3_3.1.7-3precise-20150213-1403-+0000_amd64.deb) ...
Setting up ros-hydro-openhrp3 (3.1.7-3precise-20150213-1403-+0000) ...
Processing triggers for libc-bin ...
ldconfig deferred processing now taking place
k-okada@ubuntu:~$ ls -al
/opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/longfloor.wrl
-rw-r--r-- 1 root root 4919  2月 13 03:16
/opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/longfloor.wrl
```

one tricky parts here is we uses symlink for share / sample directory
```
$ ls -al /opt/ros/hydro/share/openhrp3/share/
total 8
drwxr-xr-x 2 root root 4096  3月 13 23:18 .
drwxr-xr-x 4 root root 4096  3月 13 23:18 ..
lrwxrwxrwx 1 root root   17  2月 13 06:11 OpenHRP-3.1 -> ../../OpenHRP-3.1
```

we also could not see failure on travis,
https://travis-ci.org/tork-a/rtmros_nextage

130s, 2015-03-23:
??? I don't find anything inside of OpenHRP-3.1 directory...

```
$ sudo apt-get install --reinstall ros-hydro-openhrp3
:
Get:1 http://packages.ros.org/ros-shadow-fixed/ubuntu/ precise/main ros-hydro-openhrp3 amd64 3.1.7-3precise-20150213-1403-+0000 [7,843 kB]
Fetched 7,843 kB in 21s (364 kB/s)                                                                             
(Reading database ... 572201 files and directories currently installed.)
Preparing to replace ros-hydro-openhrp3 3.1.7-3precise-20150213-1403-+0000 (using .../ros-hydro-openhrp3_3.1.7-3precise-20150213-1403-+0000_amd64.deb) ...
Unpacking replacement ros-hydro-openhrp3 ...
Setting up ros-hydro-openhrp3 (3.1.7-3precise-20150213-1403-+0000) ...
Processing triggers for libc-bin ...
ldconfig deferred processing now taking place
$ 
$ ls -al /opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/
ls: cannot access /opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/model/: No such file or directory
$ ls -al /opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/
ls: cannot access /opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/sample/: No such file or directory
$ ls -al /opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1/
total 8
drwxr-xr-x 2 root root 4096 Mar 12 06:57 .
drwxr-xr-x 3 root root 4096 Mar 10 17:04 ..
$ ls -al /opt/ros/hydro/share/openhrp3/share/
total 12
drwxr-xr-x 3 root root 4096 Mar 10 17:04 .
drwxr-xr-x 4 root root 4096 Mar 23 14:46 ..
drwxr-xr-x 2 root root 4096 Mar 12 06:57 OpenHRP-3.1
```

k-okada, 2015-03-23:
please try `rm -fr /opr/ros/hydro/share/openhrp3` before rosinstall
or try
```
$ wget 
http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-hydro-openhrp3/ros-hydro-openhrp3_3.1.7-3precise-20150213-1403-+0000_amd64.deb
$ dpkg -c ./ros-hydro-openhrp3_3.1.7-3precise-20150213-1403-+0000_amd64.deb 
 | grep share/openhrp3/share
drwxr-xr-x root/root         0 2015-02-13 23:11 
./opt/ros/hydro/share/openhrp3/share/
lrwxrwxrwx root/root         0 2015-02-13 23:11 
./opt/ros/hydro/share/openhrp3/share/OpenHRP-3.1 -> ../../OpenHRP-3
```

130s, 2015-03-28:
> please try `rm -fr /opt/ros/hydro/share/openhrp3` before rosinstall 

After this, simply installing via apt-get worked. Strange though.

closed 2015-03-28 by 130s