start-jsk/hrpsys #141: failed to build hrpsys using jsk.rosbuild

furushchev, 2015-06-15:
```bash
bash /tmp/jsk.rosbuild -s --rtm --from-source hydro -y
```
をするとコンパイルがhrpsysで止まってしまいます。

```bash
cd ros/hydro_parent
catkin clean -a
# 新しいターミナル
source /opt/ros/hydro/setup.bash
cd ros/hydro_parent/src
bash jsk-ros-pkg-jsk_roseus-setup.sh
bash rtm-ros-robotics-rtmros_common-setup_upstream.sh
catkin b hrpsys
```

をしてもhrpsysでfailします。

ビルド時のログを以下に貼ります。何が原因でしょうか。

https://gist.github.com/6ecae2fc7684702a9dcc

k-okada, 2015-06-15:
1. this repo is old,

2. goto directory under build and make VERBOSE=1 to find the problem

snozawa, 2015-06-15:
今はhrpsys-baseをhydro_parentにもってきてビルドしているので、
このリポジトリはdeprecatedになってると思います。
おそらく、本件に関する報告は
https://github.com/jsk-ros-pkg/jsk_common/issues
がよさそうです。

closed 2015-06-15 by k-okada

furushchev, 2015-06-15:
@k-okada @snozawa 了解しました。　`goto directory under build and make VERBOSE=1`してからissueをもう一度立ててみようと思います。
ありがとうございましま。