start-jsk/hrpsys #142: このパッケージとhrpsys-baseとの関係

ishiguroJSK, 2016-04-14:
hrpsys/samplesのreadmeのチュートリアルのように
`rosrun hrpsys samplerobot-stabilizer.py`
と実行しようとして「samplerobot-stabilizer.py」がないとおこられ，
`rosrun hrpsys samplerobot_stabilizer.py`
とハイフンからアンダースコアにしたら実行できたのであまり気にかけなかったのですが，
このgitレポジトリには確かに「samplerobot-stabilizer.py」があるように見えます．
なんとなく(http://github.com/fkanehiro/hrpsys-base)のパッチ(？)かラッパー(？)のような気がしているのですが，このパッケージはどう扱うのが正しいのでしょうか？
すでに(http://github.com/fkanehiro/hrpsys-base)から落とした~/ros/indigo_parent/src/hrpsysが存在する場合，このパッケージをどのように導入するのでしょうか？

snozawa, 2016-04-14:
実はこのパッケージは不要なのでインストールもいらないです。
現状の~/ros/indigo_parent/src/hrpsysにhrpsys-base本家のソースをいれて、
catkinやROSの仕組みでつかえるようにする前の名残りがこのパッケージになっています。

> rosrun hrpsys samplerobot-stabilizer.py

ハイフンをいれているドキュメントはどこかにありますか？
今はアンダーバーでOKなので、ドキュメントを直す必要があります。

ishiguroJSK, 2016-04-14:
[https://github.com/start-jsk/hrpsys/tree/master/samples](url)です．
すべて「_」で統一ですね．わかりました．
また，例のごとく，ST関連のチュートリアルは
`rtmlaunch hrpsys samplerobot.launch`
を
`rtmlaunch hrpsys samplerobot.launch TORQUE_CONTROL:=true`
などとしないと
> 1. getParameter
>   getParameter() => OK
> 2. setParameter
>   setParameter() => OK True
> 3. start and stop TPCC st
>   This sample is neglected in High-gain mode simulation
> 4. start and stop EEFMQP st
>   This sample is neglected in High-gain mode simulation
> 5. EEFMQP st + SequencePlayer loadPattern
>   This sample is neglected in High-gain mode simulation
> 6. EEFMQP st + Turn walk
>   This sample is neglected in High-gain mode simulation
となって最後まで実行されないようなので直してもいいですか？

snozawa, 2016-04-14:
hrpsys-baseのsampleのREADMEページみるとあってない？
https://github.com/fkanehiro/hrpsys-base/tree/master/sample/SampleRobot

ishiguroJSK, 2016-04-14:
本家のREADMEを参照すればよかったのですね．
確かにきれいに答え合わせのようになっていますね・・・
ありがとうございました．

closed 2016-04-14 by ishiguroJSK

snozawa, 2016-04-14:
READMEにもこのリポジトリはdeprecatedなのでhrpsys-baseを参照してくださいとの旨かきました。
https://github.com/start-jsk/hrpsys/pull/143